remcom/hass-victron #3: Feature: add Support for New "Max System import/Export current"

pos-ei-don, 2025-01-05:
I need this Feature so that NY Main fuse from the house (which is limiter to 28A per phase) does not trigger when I connect 2 ev cars to the wallboxes. This victron Feature works perfectly well

remcom, 2025-01-06:
I'm planning to look in to this! I want to check al new registers that the Cerbo supports that are not implemented yet

pos-ei-don, 2025-01-06:
Thank you! Just out of curiocity: do everything  has to be done with modbus, or Might it be possible to set some special Features via mqtt?
I think, some Features Might be easyer to implementiert mixing modbus and mqtt

remcom, 2025-01-07:
Probably not. I’m not sure why Modbus was chosen over MQTT in the past, but perhaps we can make use of it in the future.