remcom/hass-victron #11: Additional EV Charger Station (EVCS) states

rolfcoder, 2025-01-06:
Hi,
latest [Modbus-TCP register list for the EV Charger Station (EVCS)](https://www.victronenergy.nl/upload/documents/EVCS-Modbus-TCP-register-list-v3.4.xlsx) is version 3.4 and it contains more statuses for the Charger status (address 5015) like CHARGING_LIMIT, START_CHARGING and STOP_CHARGING. 
Can you please add the following statuses under class evcharger_status(Enum) in const.py ?
    RESERVED15 = 15
    RESERVED16 = 16
    RESERVED17 = 17
    RESERVED18 = 18
    RESERVED19 = 19    
    CHARGING_LIMIT = 20
    START_CHARGING = 21
    SWITCHING_TO_3_PHASE = 22
    SWITCHING_TO_1_PHASE = 23
    STOP_CHARGING = 24

I have used these myself for couple of months already (on sfstar/hass-victron) and it prevents "NONDECODABLE" statuses when using the victron charging station.
It would be great when these are added before I switch over to remcom/hass-victron.
I'm currently on latest firmware versions for EVCS NS (v2.0) and Cerbo GX (v3.52)

remcom, 2025-01-06:
There added in the latest release!

rolfcoder, 2025-01-06:
Thanks!
Just followed the migration steps to switch to this repo: works !

closed 2025-01-06 by rolfcoder