remcom/hass-victron #8: Feature: add support for several hubs

lsgv, 2025-01-06:
Thanks for picking up work on this integration. I have 3 hubs in my installation (3 separate distant physical locations each with its own CCGX or Cerbo) and have been "hacking" the integration to appear 2 more times by copying it into a new directory and changing "domain" and "name" in manifest.json and "domain" in const.py. I don't know python so "trialed and errored" until I got it to work. It's all OK now but just in case other users out there would need this feature...

Mehh0, 2025-01-08:
This is the current message I'm getting from the gateway 
[image]

This is what I'm getting from home assistant
[image]

Ignore the IP address different as this is done on remote site towers and we have internal network routing that I can ping everything. I have 3 hubs already added with all in range from 172.17.5.0/24 172.20.3.0/24 172.16.4.0/24

lsgv, 2025-01-08:
Ok so clarification of my feature request. Currently it is possible to add several hubs (apparently, up to 3 (as I have 3 I didn't bump into the problem of adding a fourth) but, as Victron device identifier is fixed for system (100) and Venus (227) sensors from different hubs have the same name and raise Home Assistant errors ("Platform does not generate unique IDs .....") and cannot be used as such.
As it is possible to rename a hub, the actual request would be to concatenate the hub name into the sensor name so we have unique sensor names. I don't know python so cannot figure out how to do it but, and this is an uninformed opinion, I guess it shouldn't be too difficult.

intelliIT, 2025-02-25:
closed mine.
is this on the roadmap though?