remcom/hass-victron #13: Any idea how to integrate battery charge / discharge into the HA energy graph?

RSickenberg, 2025-01-07:
Hi, 

Thanks for the fork and the work done yet, I'm re-trying this integration despite the cleanup needed to each sensors / options available. I'm here to support if needed with my solar / battery / charger installation.

I'm here now to "just" have an integration about the battery energy status who might be "wrong" or differently set.

[image]
[image]

I know what's the difference between a `W` and a `Wh` but I'm not an expert into the modusTCP available on all victron here.

[image]

> Here is my devices

RSickenberg, 2025-01-07:
[home-assistant_victron_2025-01-07T12-34-20.338Z.log](https://github.com/user-attachments/files/18332554/home-assistant_victron_2025-01-07T12-34-20.338Z.log)
Also, here a debug attachment if this can help :)

remcom, 2025-01-07:
@RSickenberg Those sensors are not always working and depending if the BMS supports this. In your Case with the Pylontech BMS doesnt support this. I had the same (also using pylontech). But there is a solution for this using HA helpers.

- first make a template sensor helper Battery Charge with the template `{{ max(0, states('sensor.victron_system_battery_power_2') | float) }}` and unit watt
- make a template sensor helper Battery Discharge with the template `{{ max(0, 0 - states('sensor.victron_system_battery_power_2') | float) }}` and unit watt

now we have to instances for charge and discharge power 

- now make another helper of type integral sensor.\
name: Battery Charge kWh\
Time unit: Hours\
input sensor: select helper u made for Battery Charge
integration methode: Left Riemann sum
precision: 2

The same can be done for discharge. These 2 entity you can now use in the Energy dashboard

mauricevanderlee, 2025-01-07:
In my system with the Pylontech batteries the battery_history is also not working. I am using vebus sensors:

- Battery Charge:  sensor.victron_vebus_acin1toinverter
- Battery Discharge: sensor.victron_vebus_invertertoacout

I made your sensors to and start to compare.

DJMAXIZA, 2025-01-07:
I cant speak for the specific batteries mentioned above. One thing that could be preventing this (the case with my Fredom Won batteries), is that Victron GX devices (registers), does not support getting kWh from battery BMS that is connected via CANbus.

I have been hoping that Victron would develop this ability over time, but nothing as yet. Therefore no sensor data to feed the Energy dashboard with kWh data, for Battery Charged Energy and Discharged Energy.

It should be relatively simple coding in this integration, to use the data of V, A, etc, to produce new sensors with such data? I am hoping this could be something the owners of this integration can take on as a new feature?

remcom, 2025-01-07:
@mauricevanderlee You can indeed use that sensor but that will only work when you 
- dont use the AC out of the Multi to power devices.
- When you draw dc power in the case of camper/boats
- If you have MPPT charger on your DC bus

@DJMAXIZA I actually never thought of this. That is a good idea to look in to. I will put it on the list

kcoffau, 2025-01-07:
This is my Config:
[image]

Helper export:
[image]
[image]

Helper import
[image]
[image]

Both of those are getting Templates

  - sensor:
      - name: Battery Import Export
        unique_id: battery_import_export
        state:  >
          {% set battstate = states('sensor.victron_battery_power_system') %}
          {% if is_number(battstate) and battstate | float > 20 %}
            Import
          {% else %}
          {% if is_number(battstate) and battstate | float < -20 %}
            Export
          {% else %}
            Neutral
          {% endif %}
          {% endif %}
  - sensor:
      - name: Battery Export
        unique_id: battery_export
        state_class: measurement
        device_class: power
        unit_of_measurement: 'W'
        state:  >
          {% if is_state('sensor.battery_import_export','Export') %}
          {{ states('sensor.victron_battery_power_system') | float(0) * -1 }}
          {% else %}
            0
          {% endif %}
  - sensor:
      - name: Battery Import
        unique_id: battery_import
        state_class: measurement
        device_class: power
        unit_of_measurement: 'W'
        state:  >
          {% if is_state('sensor.battery_import_export','Import') %}
          {{ states('sensor.victron_battery_power_system') | float(0) }}
          {% else %}
            0
          {% endif %}

Probably an easier way, but I did this before the Intergation, and I was Modbus TCP direct.

dazza120, 2025-01-08:
@remcom Oh my word i assumed the pylons could not do it all this time my energy chart is rubbish without it but you guys have my head spinning go to template make a helper template code {{ max(0, states('sensor.victron_system_battery_power_2') | float) }} use watts as unit, but is there no device class or anything is it just those two things and the name of the helper?

remcom, 2025-01-08:
Oh. forgot about those. Device class: power and state class: Measurement

dazza120, 2025-01-08:
@remcom the integral sensor metric prefix are we using Killowats?

remcom, 2025-01-08:
that should be Kilowatthours (kWh).

dazza120, 2025-01-08:
@remcom that bit only comes as a metric Killo Mega Giga Tera i assume because you use time in units thats what chages it to kWh

remcom, 2025-01-08:
This is what it need to be. I did it out of the top of my head but i was wrong. Set your precision on 2 or 3. not 1 as the screenshot

[image]

dazza120, 2025-01-08:
> wrong. Set your precision on 2 or 3. not 0 as the screenshot

Done thanks am I able to use that code for my Shelly to do the grid from
that and call it I the code sensor.shelly_em_power_2 ? And do the same
thing with the helpers as I don’t think the grid stuff is reading correctly
from Octopus’s end

fyrstyk, 2025-01-08:
Have you looked at the dbus to see what data is coming off the battery bms?

dazza120, 2025-01-08:
Yeah thanks but the pylontec’s can’t do it there BMS doesn’t do it but I
followed the guide now it’s working is it possible I can use the same thing
with my Shelly EM to get the grid as for some reason it won’t let me add
the entity in the energy data bit

fyrstyk, 2025-01-11:
> Yeah thanks but the pylontec’s can’t do it there BMS doesn’t do it but I followed the guide now it’s working is it possible I can use the same thing with my Shelly EM to get the grid as for some reason it won’t let me add the entity in the energy data bit

I'm using  sensor.victron_vebus_acin1toacout_227 for import and victron_vebus_invertertoacin1_227 for export.