remcom/hass-victron #16: What Test is needed to remove "This integration wasn't tested with a three phase system" from readme?

pos-ei-don, 2025-01-08:
I use it heavily with a three phase system. What Tests should i perform so we can remove this?
I think, it works nicely and i am not aware of a problem related to phases

Raini20, 2025-01-08:
I second this. I use this integration since the beginning and never had a problem with 3-phase.

breuerm710, 2025-01-08:
Same for me, the transition worked w/o any issue on my three phase system.
Cerbo on 3.5.2
Multi on 5.1.0
MPPT 250/85 on 3.16
VM-3P75CT on 1.08
openDTU for hoymiles WR

Handy-Harry, 2025-01-08:
Yup, 3 phase system here. No issues.

remcom, 2025-01-08:
I think its enough that you are already reporting it that its working fine. I put a label on this issue and i will update the documentation next weekend

remcom, 2025-01-09:
Documentation updated in the latest release. Thanks for the input!

closed 2025-01-09 by remcom