remcom/hass-victron #17: Migration failed: {"message":"Invalid handler specified"}

john-trn, 2025-01-08:
Hi, the migration failed on my side.

I removed the old “Victron GX Modbus TCP” integration in HACS and downloaded the new one.
Then I restarted but the Victron integration is gone and when I try to add the integration in "Devices and Services" I get the following error message:
"Fehler

Der Konfigurationsfluss konnte nicht geladen werden: {"message":"Invalid handler specified"}"

System Log:

Logger: homeassistant.config_entries
Quelle: config_entries.py:3314
Erstmals aufgetreten: 09:04:21 (3 Vorkommnisse)
Zuletzt protokolliert: 09:11:22
Error occurred loading flow for integration victron: No module named 'pymodbus.pdu.register_read_message'

Do you guys have a hint for me what to do?

RrobertRr, 2025-01-08:
Same issue for me

dmsims, 2025-01-08:
Did you download in Step5 ?

From the migration guide:

1 Ensure you first upgrade to the latest version of Home Assistant (2025.1.0 or higher). After this upgrade, the Victron integration will stop working, but that’s fine for now.
2 Go to HACS and remove the “Victron GX Modbus TCP” integration.
When you attempt this, a message like the following may appear:
“The Victron GX Modbus TCP integration is configured or ignored. You need to delete the configuration for it before removing it from HACS.”
You can safely click Ignore to proceed.
3 Add the new repository using this link: Open your Home Assistant instance and open a repository inside the Home Assistant Community Store.
4 Search for “Victron” in HACS.
You will find two Victron integrations. Select the new one. You can verify it by checking that the documentation starts with:
“This Repo is a fork of https://github.com/sfstar/hass-victron and is a continuation of that project.”
5 Download the integration.
6 Restart your Home Assistant instance.
After restarting, the Victron integration should work as expected.

RrobertRr, 2025-01-08:
> Did you download in Step5 ?
>
> From the migration guide:
>
> 1 Ensure you first upgrade to the latest version of Home Assistant (2025.1.0 or higher). After this upgrade, the Victron integration will stop [...]

=> Yes, I installed the correct integration (the new one)

john-trn, 2025-01-08:
Yes, me too. Deleting the Victron Integration does not work either (to start all over again).

remcom, 2025-01-08:
@john-trn @RrobertRr Make sure you run HA version 2025.01.0 or higher. This will happen by a mismatch in the installed pymodbus version. This should by 3.7.4. This error happens if your using a pymodbus versions 3.6 or lower or 3.8 and higher.

john-trn, 2025-01-09:
@remcom I'm using:

```
Core 2025.1.1
Supervisor 2024.12.3
Operating System 14.1
Frontend 20250106.0
```

 and therefore I think
```
# homeassistant.components.modbus
pymodbus==3.8.3
```

It's the ova template so the system runs as a virtual machine on top of a esxi host. Never had issues with that before.

remcom, 2025-01-09:
@john-trn Latest release is using 3.7.4: https://github.com/home-assistant/core/blob/d59a91a905e4a0e0ed0dc19f5da17529b2f3da84/requirements_all.txt#L2097

Only the HA dev branche is updated to 3.8.3. So when that is in a release than i need a make s mall update for this. I was already aware of this. I can make a PR already for this that you could use

remcom, 2025-01-09:
I did some quick research and maybe i can make a version that is compatible with both pymodbus packages. I will make a fix for this this evening

remcom, 2025-01-09:
@john-trn or @RrobertRr Are you able to test this change and check if its working with your pymodbus version? https://github.com/remcom/hass-victron/pull/22/files

john-trn, 2025-01-09:
@remcom Thanks for your very fast reaction!

I can now add the Victron integration but cannot connect to my Venus Device.
This is the error in HA:

```
Logger: custom_components.victron.hub
Quelle: custom_components/victron/hub.py:84
Integration: victron ([Dokumentation](https://github.com/remcom/hass-victron), [Probleme](https://github.com/remcom/hass-victron/issues))
Erstmals aufgetreten: 10:46:11 (10356 Vorkommnisse)
Zuletzt protokolliert: 11:06:46

    ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given
    ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
```

remcom, 2025-01-09:
I will have a look later today. I hoped the functions stayed the same with the new version and just a import from a different location ;p

remcom, 2025-01-09:
@john-trn I made some extra changes. Can you test with does. I cant easily test it myself because i cant easily update my test HA yet to use the new pymodbus so thnx for the testing

john-trn, 2025-01-09:
@remcom Yesss, perfect. Now it is all working again! Thank you so much!

remcom, 2025-01-09:
Nice! When i'm back home i will make a release with this fix

remcom, 2025-01-09:
Fix is now available in latest release. Thanks for the input!

closed 2025-01-09 by remcom

RrobertRr, 2025-01-11:
For me it is also working now.
Thank you a lot @remcom .