remcom/hass-victron #18: Cerbo/Venus OS version 3.50

kucerazdenek, 2025-01-08:
Hi there,

this is not an issue as such, just wanted to ask if anyone has experience with running this integration on v3.50? I am currently on v 3.40 and would like to upgrade to 3.50

Thanks

bidouilleur31, 2025-01-08:
Can this not be discussed in the discussion section instead ?

remcom, 2025-01-08:
Hi @kucerazdenek Its good to make a new discussion about this. https://github.com/remcom/hass-victron/discussions

I'm running on 3.50 without any issue

trantoriana, 2025-01-08:
Here is a person running 3.5.2
Perhaps consult that person?

https://github.com/remcom/hass-victron/issues/16#issuecomment-2578151849

T

fyrstyk, 2025-01-08:
Running 3.53 here and works fine.

kucerazdenek, 2025-01-09:
Sorry guys to not placing this into discussion and thanks for confirmation that integration works with new version well!!

closed 2025-01-09 by kucerazdenek