remcom/hass-victron #20: Victron EVCS (EV Charging Station) the sliders for Maxcurrent only has Values 0 and 1 instead of 10A to 32A

DeeSaR47, 2025-01-08:
Hi Remco,

in the controls section for the Victron EVCS (EV Charging Station) the sliders for Maxcurrent and Setcurrent only have the values 0 and 1 instead of values between 10A to 32A for Maxcurrent respectively 6A to 32A for Setcurrent.
[image]

Also some sensors for this (and other) values are unavailable.

[image]

I dont know if this is fixable without access to a EVCS unit but I would be glad to help testing if required.

Thank you for your work.

remcom, 2025-01-08:
Hi @DeeSaR47 I put it on my list to have a look at it. I dont have an EVCS so i will let you know if i have something you can test!

jpouyaud-perso, 2025-01-28:
hi,
What I don't understand is this was working before the fork
for 1 to 32A
cheers

jpouyaud-perso, 2025-01-28:
correction
I was able to find back the value from 1 to 32A by setting ac_current to 32 in the integration/configuration

DeeSaR47, 2025-01-29:
Hi jpouyaud-perso,
no the setting for the manual charging current from 10A to 32A and the setting for the maximum allowed charging current in increments of 1A also wasn´t working before the fork.
In Vicrton VRM, the Remote Console of cerbo and the chargers own webui this is working fine.

jpouyaud-perso, 2025-01-29:
what I can say is once I've installed the Remco fork, I've got the same issue than you
ac_current* was set to 1

I've changed it to 27

[image]

and then I've got the range from 1 to 27A for my EVCS

[image]

I think this ac_current setting is just not well referenced, but this is the value we need to set for EVCS

cheers