remcom/hass-victron #25: All Victron entities are flagged as unavailable but the data is available and updating

adrian-schutte, 2025-01-10:
Hi,

After the switch to this repo all the Victron entities are flagged as Unavailable but the data is available and updating.

The 6 steps in the migration guide were followed to switch to this repo.

Did something go wrong with the switchover?

Here is an example of one of my solar chargers.

[image]

remcom, 2025-01-10:
Are all the devices in the Victron integration unavailable? What kind of errors do you see in the HA log? For id 0 devices this can be the case that some entities are not wirking there anymore. I would not expact this for other devices

adrian-schutte, 2025-01-10:
Hi Remco,

Yes it is for all the Entities of the Victron Devices.

Attached are debug logs of the Victron Integration.

adrian-schutte, 2025-01-10:
Here they are:

[home-assistant_victron_2025-01-09T22-29-36.347Z.log](https://github.com/user-attachments/files/18376790/home-assistant_victron_2025-01-09T22-29-36.347Z.log)
[home-assistant_victron_2025-01-09T22-41-13.498Z.log](https://github.com/user-attachments/files/18376791/home-assistant_victron_2025-01-09T22-41-13.498Z.log)

remcom, 2025-01-10:
I dont see any issues in the logs your provided.

I do see a entity called 'sensor.poolsense_undefinedtype_singleton_5' thats throwing a bunch of issues

For the Victron part averythin lokgs as expected it find  (100, 227). that matches with the available devices you shared in your discussion. Not sure why the gps part isnt showing up. I can look into that.

```
2025-01-10 14:41:58.853 DEBUG (MainThread) [custom_components.victron.config_flow] successfully discovered devices
2025-01-10 14:41:58.871 DEBUG (MainThread) [custom_components.victron.coordinator] Fetching victron data
2025-01-10 14:41:58.871 DEBUG (MainThread) [custom_components.victron.coordinator] {100: ['settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], 227: ['vebus_registers']}
2025-01-10 14:41:58.968 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.097 seconds (success: True)
```

Did you try a full restart and after that checked these devices?

adrian-schutte, 2025-01-10:
I have tried multiple restarts, also went through the 6 steps again without any success.

On the GPS discussion that was started by [Accuvue26](https://github.com/Accuvue26) I thought let me test it from my side as I have a test Cerbo GX and USB GPS lying around.

remcom, 2025-01-10:
Ah. didn't notice that. my bad. I will have a look at the GPS part later. Big change that it wasnt working for a long time and nobody noticed it.  

And if you open id 100 in HA what do you see? The logs show that all the data is fetched without any issues for id 100 and 227. Can you share a screen shot?

adrian-schutte, 2025-01-10:
For this issue #25 the problem is that all the Victron Entities for my production Cerbo GX are showing that they are Unavailable but data is being fetched and stored in the History of HA for all these Entities as is shown in the example screen capture above.

Issue #26 is for the problem where my Energy Meter on my production Cerbo GX has no updated data in HA since the move to the new repo. You have indicated that this is related to issue #7.

So for #25 the rescan is not an issue it seems to be more of a cosmetic issue where all the Victron Entities are shown as Unavailable in HA

remcom, 2025-01-10:
for issue #26 i would expect that the grid meter now should show op in id 100 with the entities. Can you try the previous release 0.3.3 and see if the grid data shows up in id 100

For this issue. Kind of weird that the data is logged but not showing up correctly in the interface. Not sure what to do for this or whats causing this

adrian-schutte, 2025-01-10:
#26 I originally did the switchover with v0.3.3. I then later updated to v0.3.4 but the grid issue persisted. I have now downgraded again to v0.3.3. Updated data still not showing for id 0 and no new entries in id 100. I also did a resync again. Same for id 0 and 100 as before the resync.

Here is a debug log for the resync on v0.3.3.

[home-assistant_victron_2025-01-10T13-35-04.430Z.log](https://github.com/user-attachments/files/18378977/home-assistant_victron_2025-01-10T13-35-04.430Z.log)

I did a filtered search on Entities and only the Entities for id 0 and 31 show up. Nothing for 100. 31 is not used any more it was a Energy Meter for AC Loads.

[image]

If I click on "grid energy forward total" from above I get the following:

[image]

 If I then click on Show more I get this:

[image]

The following modbus addresses which were previously written to Entities in id 0 are now nowhere to be found except for in id 0 but they are not updated since the first switchover.

[image]

#25 I have upgraded to 2025.1.2 and connected with incognito browser and the Victron Entities are now showing data. Also fine with normal browser now. Not sure if it was browser or 2025.1.2 issue that solved this.

I am running two HA environments. The prod Cerbo is in prod HA and the test Cerbo is in test HA

remcom, 2025-01-10:
@adrian-schutte Nice that this ticket is solved at least. I dont have the time tonight to look further in issue #26 . I will dig into this a little further tomorrow and i will get back to you!

remcom, 2025-01-10:
Could not let it go so i did a little digging.

Are you able to change files on you HA instance to test something?

In the custom_components/victron/hub.py Can you comment/delete line 70 and 71. Then restart the HA instance and then to a rescan. I'm curious if the grid meter now shows ip in id 100

[image]

adrian-schutte, 2025-01-10:
I made the changes to the hub.py file on my test HA. After the rescan the GPS info is there on id 100. See below.

I will now make the same change on my prod HA and see if the grid meter also appears.

[image]

adrian-schutte, 2025-01-10:
My test HA is on v0.3.4. and prod HA is still on v0.3.3 after the downgrade earlier today. I see the hub.py differs between the two versions. Should I rather update my prod HA to v0.3.4 before making the change?

remcom, 2025-01-10:
It doesn't really matter on which version you test. If you test with 0.3.3 please find the same lines. Probably a little higher ow lower but they are the same between the versions.

adrian-schutte, 2025-01-10:
It is lines 65 and 66 on 0.3.3.

The grid Entities are now under id 100 on prod HA after editing hub.py, restarting and rescanning.

[image]

Here is an example grid entity with Entity id sensor.victron_grid_energy_forward_total_2. There are now three similar entries compared to the previous two.

[image]

remcom, 2025-01-11:
Nice that there showing now. Is the gps also showing up by any change? The other entities are probably when this entity was connected to a different device (old id 0)

There is also a beta release now with this fix. You can download it with hacs if you open this integration and then in the menu in the upper right corner and click redownload. then select the latest beta version

adrian-schutte, 2025-01-11:
I have no GPS device connected to my prod Cerbo but on the test Cerbo it shows up under id 100 as per screen capture here https://github.com/remcom/hass-victron/issues/25#issuecomment-2583336862 after commenting out the two lines in hub.py, restarting HA and rescanning.

I have installed v0.3.5-beta.1 on my test HA restarted and rescanned. The GPS Entities are still under id 100 and updating.

[image]

[image]

[image]

remcom, 2025-01-11:
That they show up under id 100 is as expected. All devices. The integration cant distinguish multiple devices on the same id. A work round is that you move the gps device to a different id in VRM.

In the future i would love if that every device on the same id is also getting there own device in HA but this is not the case yet

adrian-schutte, 2025-01-11:
I have now installed v0.3.5-beta.1 on prod HA as well. Grid Meter still under id 100 and updating after restarting and rescanning.

[image]

[image]

remcom, 2025-01-14:
@adrian-schutte I think this issue is solved now in the latest release?

adrian-schutte, 2025-01-16:
@remcom Yes updated the integration to v0.3.5 on test and prod HA and both working as expected. Thank you very much for resolving this issue.

closed 2025-01-16 by remcom