remcom/hass-victron #26: Data for Victron Energy Meter VM-3P75CT connected to VE.Can not available

adrian-schutte, 2025-01-10:
Hi,

After the switch to this repo data for the Victron Energy Meter VM-3P75CT  is not available anymore.

The meter is connected to the VE.Can port on the Cerbo GX and is configured as follows:

[image]

It is showing up as follows under the Cerbo GX's "Services" "Modbus TCP" "Available Services" screen.

[image]

The 6 steps in the migration guide were followed to switch to this repo and after I realised the Energy Meter is not updating any of its entities with data I ran a rescan.

Is the below my issue?

> Breaking change: Users that have a device with modbus id 0 connected should perform a rescan after installing this update. This will seperate the modbus device with id 0 into it's seperate dedicated device. Unfortunately historical data for the entities, when the device was placed under unit id 100, will not be transfered. All other entities and devices should remain unaffected.

This Device and its Entities was available before the repo switch as per the screen captures below.

[image]

[image]

[image]

I do not see any new Device or Entities that was created after the rescan for this Energy Meter.

remcom, 2025-01-10:
I think this is a duplicate of your own ticket: https://github.com/remcom/hass-victron/issues/25

Maybe beter to close this one so we can discuss every thing in your other ticket?

adrian-schutte, 2025-01-10:
Hi Remco,

This is not a duplicate ticket. On this Victron Device there is no new data
being logged for any of its Entities since the switch to the new repo. With
the other ticket all the other Entities of the Victron Devices are still
logging new data but they are being flagged as Unavailable.

remcom, 2025-01-10:
Then this is indeed the breaking change issue you found. This ticket is a duplicate of https://github.com/remcom/hass-victron/issues/7

remcom, 2025-01-16:
@adrian-schutte I think the issue for this ticket are also resolved?

adrian-schutte, 2025-01-16:
@remcom That is correct. We actually tracked the resolution of this issue in #25.

closed 2025-01-16 by remcom