remcom/hass-victron #33: ESS Mode select not working

fyrstyk, 2025-01-11:
I have been trying to change my ESS mode from the integration, however no matter what mode I select there is no change on the GX device.

Any ideas?

VenusOS 3.53
Integration version 0.34

[image]

remcom, 2025-01-11:
@fyrstyk I was actually looking into that today. I never used the write mode before and did some testing today. I also noticed a few other options that didnt work or sometimes worked and other times it didn't. Not sure yet why.

dmsims, 2025-01-11:
Does it work if you try an automation ?

[image]

fyrstyk, 2025-01-11:
@dmsims  I've not tried an automation.

@remcom I've written a temporary workaround in nodered and have used MQTT for the time being to change mode.

We heavily customise the UI and also control over MQTT on GX devices at work and I was wondering why Modbus is used instead of MQTT?