remcom/hass-victron #34: can't delete old data with id 0

testpaul999, 2025-01-13:
Hi, 
first thx for this fork! :-)
The migartion works like a charme.

from the old integration i have some old data "settings" id 0 and cant delet them.
"the integartion must stop deliver them" or so...
[image]

greating and thx for help.

remcom, 2025-01-14:
Hi @testpaul999 . Thanks for mentioning this. I will put it on ly list!

maps0n, 2025-03-24:
Thank you for your work.
I am facing the same issue:
There is device settings mit model number 100 and there is second one, settings-0 with model number 0:
Both contain the same set of entities.
Anything I can do, or do I have to wait for an update?
Additional question: I first assume the model number corresponds to the VRM instance number. But they don't match.