remcom/hass-victron #35: Problem with installation : no entity

chanshaillan, 2025-01-14:
victron HACS installation 0.3.5
victron installation integration 
add configuration

but no entity

it s first installation. i put adresse ip of my ekranos GX and port 502...

any idée

regards

remcom, 2025-01-14:
check you enabled modbus in your GX device. You can find this in your VRM -> Remote console -> settings ->services and make sure to enable modbus TCP.

After this try to add it again. If stil no devices showing make provide a log from home assistant with the output of this integration

chanshaillan, 2025-01-14:
modbus was and is enable

Enregistreur: pymodbus.logging
Source: custom_components/victron/hub.py:24
intégration: victron (documentation, problèmes)
S'est produit pour la première fois: 19:19:25 (34 occurrences)
Dernier enregistrement: 20:55:47

Connection to (192.168.1.84, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.82, 502) failed: [Errno 111] Connection refused
Connection to (http://192.168.1.84/gui-v2/, 502) failed: [Errno -2] Name
does not resolve
Connection to (http://192.168.1.84, 502) failed: [Errno -2] Name does not
resolve

chanshaillan, 2025-01-14:
connection refused.
but in chrome when write ip : http://192.168.1.84  i can go on vrm portal

remcom, 2025-01-14:
Only fill in the ip address. i also see that you tried including http://. With only the ip i see `Connection to (192.168.1.84, 502) failed: [Errno 111] Connection refused` You need to check if your HA instance also have access to that ip or otherwise you need to check why your GX device is refusing the connection.

Normally if modbus is enabled in HA can reach the ip all should work

chanshaillan, 2025-01-14:
ok thanks, i will try to fond.

regards