remcom/hass-victron #40: control entities get lost

EckeFL, 2025-01-24:
Hi,

some entities get lost after some time. After a rescan, the entities are back.

All number entities in settings are affected, also vebus entities.

remcom, 2025-01-27:
oke. thats no good. Dot you have any logs from this integration when this happens?

EckeFL, 2025-01-27:
I´m sorry. 

I changed my inverter settings and deactivated the sleeping mode.
I have noticed that the pvinverter entities are also no longer displayed the morning after the start. 
The entities are not available at night when sleeping mode turned on.
Perhaps this is the trigger. I'll keep an eye on it and maybe I can reproduce it. Then of course I will make a log.

513mahoni, 2025-02-25:
same here...

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.victron.coordinator
Quelle: helpers/update_coordinator.py:380
Integration: victron (Dokumentation, Probleme)
Erstmals aufgetreten: 16:07:09 (74 Vorkommnisse)
Zuletzt protokolliert: 16:19:19

Unexpected error fetching victron data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 96, in _async_update_data
    self.parse_register_data(
    ~~~~~~~~~~~~~~~~~~~~~~~~^
        data, register_info_dict[name], unit
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ).items()
    ^
  File "/config/custom_components/victron/coordinator.py", line 125, in parse_register_data
    decoder.decode_16bit_uint(), value.scale, value.unit
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/payload.py", line 413, in decode_16bit_uint
    return unpack(fstring, handle)[0]
           ~~~~~~^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes