remcom/hass-victron #42: pymodbus.logging

larronni, 2025-02-01:
Runing Home Assistant 2025.2.0b5

Logger: pymodbus.logging
Source: custom_components/victron/coordinator.py:117
integration: victron ([documentation](https://github.com/remcom/hass-victron), [issues](https://github.com/remcom/hass-victron/issues))
First occurred: 21:46:44 (740 occurrences)
Last logged: 21:49:58

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

dk1305hb, 2025-02-06:
Same here. Config EU, single phase, advanced, Interval 4sec.

[image]

Alea1337, 2025-02-13:
> Have the same issue and it is filling up the logfile

same here

vacharadek, 2025-02-16:
The same problem

Logger: pymodbus.logging
Zdroj: custom_components/victron/coordinator.py:117
integrace: victron (dokumentace, problémy)
První výskyt: 14:33:36 (95 výskyty)
Naposledy logováno: 14:34:23

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

peregrines2, 2025-02-24:
My Logfile is also running full :(
1 phase EU installation here.

crisi-solar, 2025-03-05:
Problem noch immer auch nach Update auf v0.5.0 !

peregrines2, 2025-03-05:
Here too ... was really looking forward this update to resolve this annoying crash-issue.