remcom/hass-victron #48: Got error message "BinaryPayloadDecoder is deprecated"

han-alink, 2025-02-19:
Logger: pymodbus.logging
Source: custom_components/victron/coordinator.py:117
integration: victron ([documentation](https://github.com/remcom/hass-victron), [issues](https://github.com/remcom/hass-victron/issues))
First occurred: 04:16:04 (32709 occurrences)
Last logged: 12:45:09

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

rolfcoder, 2025-02-19:
Is same as issue#42

Geisti01, 2025-03-21:
same here and it's going to add around 200.000 messages per day into the log... so hopefully somebody would be able to take a look into it before the next update breaks everything....